OpenBuilds/OpenBuilds-CONTROL #217: Feature Request: Add X and Y max travel distance for GCode somewhere on the screen

SeanDolan, 2021-06-09:
The functionality of the Check Size feature is amazing for working out if my GCode will fit on the piece of material I have already put down on the bed of my CNC. I would like to request these maximum distances be visible somewhere on the screen after loading a GCode file.

The main use is that if I go looking for a piece of material to suit the GCode, I currently have to refer back to my original stock size in my CAD software or zoom right into the ruler guides on the preview. Which is super fiddley at times.

Could you please consider this feature and putting the minimum work piece material size somewhere big and bold (or at least easy to read)? Would really appreciate it.

petervanderwalt, 2021-06-09:
You can use a Macro for that (: 

```
if (typeof object !== 'undefined') {
    var string = ""
    if (object.userData.inch) {
        if (typeof redrawGrid === "function") { // Check if function exists, because in Mobile view it does not
            string += "Width (X): " + (object.userData.bbbox2.max.x - object.userData.bbbox2.min.x) + "inch" + " / " + "Height (Y): " + (object.userData.bbbox2.max.y - object.userData.bbbox2.min.y) + "inch";
        }
    } else {
        if (typeof redrawGrid === "function") { // Check if function exists, because in Mobile view it does not
            string += "Width (X): " + (object.userData.bbbox2.max.x - object.userData.bbbox2.min.x) + "mm" + " / " + "Height (Y): " + (object.userData.bbbox2.max.y - object.userData.bbbox2.min.y) + "mm";
        }
    }
    console.log(string)
    Metro.dialog.create({
        title: "Stock Size needed",
        clsDialog: "dark",
        content: `
           To run this job, you need a piece of stock:<hr>
           ` + string + `
       `,
        actions: [{
            caption: "Cancel",
            cls: "js-dialog-close alert",
        }]
    });
}
```

petervanderwalt, 2021-06-09:
[image]

closed 2021-06-09 by petervanderwalt