OpenBuilds/OpenBuilds-CONTROL #218: Feature Request: Dark Mode

SeanDolan, 2021-06-09:
I would really like the ability to enter a dark mode for the OpenBuilds Control Software in my browser. I have been able to replicate what I would like for the interface using a Google Chrome Extension, however the preview window doesn't look like it is able to be impacted by stylesheet changes by the extension.

I know this is not a priority but it would be awesome if there was a dark mode or even better if we could select colours for different components (e.g. preview window background, foreground and rulers).

Thank you for your time reading this request. I really appreciate your valuable time and efforts on this software.

petervanderwalt, 2021-06-09:
I'll help you with some clues on playing with the ThreeJS code, but officially we won't consider adding a dark mode.  However, as you have your CSS and after this post, you can play with the ThreeJS code too, you can wrap all that into a Javascript Macro to share with the community too: 

So, the 3D view runs on https://threejs.org/ 

Checkout https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/app/js/viewer.js and https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/app/js/viewer-ruler.js to see how the viewer is drawn

From the JS Macro, you can access the Renderer and other objects, overwrite existing functions, etc to change whatever you want: Learn more about JS Macros here: https://openbuilds.com/threads/help-getting-started-with-javascript-macros-in-control.15880/#post-97234 and also https://thayneco.com/openbuilds-control-javascript-macros-getting-started/ 

For, for example, removing the "sky" and setting the Renderer Background, allows you to make the Background of the 3D view darker:

workspace.remove(sky);
renderer.setClearColor(0x333333, 1)

Example:
[image]

Through those, into an Autorun JS macro, and it will do that on startup for you (:
You can get rid of the Chrome extension and use JQuery to setup the CSS as well: See https://www.w3schools.com/jquery/jquery_css.asp then you can use the Macro in the CONTROL UI as well. 

Closing this issue for now, but feel free to reply here with questions and sharing your work, happy to help you promote your JS Macro (;

petervanderwalt, 2021-06-09:
[image]

Wrap it in a setTimeout so that the renderer is up and ready when it executes (:

closed 2021-06-09 by petervanderwalt

petervanderwalt, 2021-06-10:
Updated:  One-liner Macro using CSS Filters:

```
$(':root').css("filter", "invert(1) hue-rotate(180deg) saturate(4)") 
```

Results in 

[image]

SeanDolan, 2021-06-11:
WOW! Above and beyond as always!  Thank you!