OpenBuilds/OpenBuilds-CONTROL #219: Check if machined is homed / G53

petervanderwalt, 2021-06-18:


petervanderwalt, 2021-06-24:
Check if machined is homed and if any G53 moves are in the GCODE, Ask to Home

petervanderwalt, 2021-12-01:
From #224: suggested by @sharmst 

> we could do some trick messages in the probe wizards and warning messages when starting a job that changes to a different WCS other than the current WCS. Maybe tie that into your planned "check if homed and for G53" feature you have planned.

petervanderwalt, 2021-12-10:
It does not, no

I was planning on setting a parameter in the 'status' object from index.js whenever $h passes through, and reset it to false on alarm, grbl init string, etc 

That way the UI can just reference the status - and will allow co-existence with jog widget (other web interfaces can send $h so best to check in index.js under node, rather than just in frontend)

sharmstr, 2021-12-10:
Okay cool.  That's what I thought.  Was also trying to work out a way to tell if 'fake homing' was being used.  This info helps.

petervanderwalt, 2021-12-10:
If there ever was a concept I want to ban to hell its fake homing.  Get the switches, just like, get the switches... its soooo cheap!

petervanderwalt, 2022-04-21:
[image]

>V1.0.318 - lets test it out in the field with G53 first

petervanderwalt, 2022-05-25:
Todo: Only popup Tip window if Homing is enabled in Grbl settings?

closed 2022-05-25 by petervanderwalt