OpenBuilds/OpenBuilds-CONTROL #222: Feature Request: Connect to networked controller (xPro V5)

southsideboats, 2021-10-19:
I'd love to be able to connect this to my xPro V5 that's connected to my home network via wifi. I can reach my controller setup by navigating to http://192.168.50.201/ for example, but OpenBuilds Control only has the option to connect via USB.... and I like it better than the built-in interface.

sharmstr, 2021-11-12:
If I remember correctly, the xPro 5 uses a different version of grbl so even if you could connect, Control still wouldnt work with it.

karoria, 2021-11-12:
In my knowledge, xPro V5 uses grbl_esp32 firmware and that is just grbl as far as communication is concerned. So no need to change anything specific for it. Yes, the network communication is to be made either by websockets or by telnet. They also use streaming, so the connection part needs to be altered only. The responses are just same as we received via usb connection (strings separated by \r\n). That said, parsing and arranging data in UI will still remain same. I can tell this because I am partially successful in writing my custom app, lot of work is still pending.

sharmstr, 2021-11-12:
Unless something had recently changed, it doesnt work with Control.  https://openbuilds.com/threads/help-with-xpro-v5-config.17000/

karoria, 2021-11-12:
Oh yes.. They have changed $ settings a lot. That may be the issue.

closed 2021-11-12 by petervanderwalt