OpenBuilds/OpenBuilds-CONTROL #224: Allow probing in other WCS systems as well

petervanderwalt, 2021-12-01:
https://openbuilds.com/threads/openbuilds-control-software.13121/page-66#post-120688

sharmstr, 2021-12-01:
Along these lines, it would be nice to display current WCS.  Maybe in the space above setzero X button.  I tried to do this myself but couldnt figure out how to get the $G info back from grbl.  Thought it might be in laststatus.

Anyhow, not a must have for sure.  But if that info was available we could do some trick messages in the probe wizards and warning messages when starting a job that changes to a different WCS other than the current WCS.   Maybe tie that into your planned "check if homed and for G53" feature you have planned.

petervanderwalt, 2021-12-01:
Updated G10 to P0 https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/55565617bb28d89cc8191d3be56a1f44812b652a up for testing as v1.0.306

petervanderwalt, 2021-12-01:
> Along these lines, it would be nice to display current WCS. Maybe in the space above setzero X button. I tried to do this myself but couldnt figure out how to get the $G info back from grbl. Thought [...]

Leaving ticket open - would be good to add it to the Parser from the backend side - I have done it for my old GrblJS host https://github.com/openhardwarecoza/grbljs/blob/master/index.js#L1090-L1258 so relatively easy to port 

In GrblJS I only request $G upon first connection, and when a G54-G59 is sent https://github.com/openhardwarecoza/grbljs/blob/master/index.js#L746-L751

sharmstr, 2021-12-01:
Awesome!  

I'll do some probe testing here shortly.  Thanks!

petervanderwalt, 2021-12-01:
Released (:

petervanderwalt, 2021-12-01:
Comments? 

Sneaked a somewhat hidden little button up here

[image]

Has a dropdown menu so you don't need to use the Console:

[image]

And the parser picks up the coordinate system from $G and updates what's shown on the button (current state)

sharmstr, 2021-12-01:
That's exactly what I was thinking.  I didnt want to push my luck by asking for it to switch WCS as well.  Whoop!

petervanderwalt, 2021-12-01:
Made it a little prettier, and added the Checkmarks so you also see in the Menu where you are

[image]

Will be out as v1.0.308 (:

Now... someday when I have a week to work on it - we gotta add 3D view of the machine+work coordinates (just colored rects showing machine size (from Grbl Max travel) and relative work offsets (:

petervanderwalt, 2021-12-02:
> In GrblJS I only request $G upon first connection, and when a G54-G59 is sent https://github.com/openhardwarecoza/grbljs/blob/master/index.js#L746-L751

This section introduced a bug, will fix today
When I wrote that for GrblJS it was still based off an older version of node-serialport - which handled the "realtime" command without interfering with the running stream - in the later versions of node-serialport used by CONTROL, it seems to

closed 2021-12-07 by petervanderwalt