OpenBuilds/OpenBuilds-CONTROL #230: Position Cone Visible Prior to Starting Job

JustinWolford, 2022-01-01:
I would like the option to see the position cone of the tool prior to starting a job in the job visualizer.

This would allow the user to verify the physical location of objects with respect to the toolpath preview. For example, making sure the tool appears to be in the correct place with respect to the origin or to verify that hold down clamps will actually be clear of the toolpath by jogging the bit to the clamp and comparing the position of the cone with the toolpath preview.

This would help the user to avoid crashing into clamps or to notice an incorrectly set origin prior to beginning the job.

petervanderwalt, 2022-04-19:
Sorry, unable to complete this request at this time (: But there is something even more fun coming in its place (;

closed 2022-04-19 by petervanderwalt

JustinWolford, 2022-07-01:
Any hints on what to be on the lookout for for this? Its keeping me from using OpenBuilds for the time being because I have some pretty complex work holding for weirdly shaped pieces. Would love to know what issue to track to know when to come check it out again. :)

petervanderwalt, 2022-07-06:
Haven't tested myself, but worth a try: create a JS macro that sets

```cone.visible=true```

petervanderwalt, 2022-07-06:
Oh no that won't work.  Will try and add a tickbox/macro option for you in next version - as the other feature went on ice for a bit

petervanderwalt, 2022-07-06:
Cone is now always-on in V1.0.324 or later