OpenBuilds/OpenBuilds-CONTROL #231: VDF via RS485

pixelmagic66, 2022-01-09:
It would be nice if a VDF could be controlled via RS485 and also receive feedback from the motor.

petervanderwalt, 2022-01-12:
Grbl 1.1 does not support RS485. See https://github.com/gnea/ 
CONTROL is just an interface application that fronts the firmware in the background

closed 2022-01-12 by petervanderwalt

pixelmagic66, 2022-01-12:
I understand that, but would it not be possible to let CONTROL communicate on itself via a RS485 dongle with the VDF ?

conucon, 2022-01-14:
Hello Peter,

 

this is Felix. I am building and selling Openbuilds in Germany. I am in Cape Town for two weeks right now. If you are around, it would be a pleasure to meet up with you.

 

Have a good day,

Felix

 

CONUCON UG (haftungsbeschränkt)

 

Billhorner Brückenstraße 40

20539 Hamburg 

Deutschland

 

Geschäftsführer: Dr. Felix Oehme

Email: ***@***.***

Tel.: +49 (0) 40 756791670 00

Mobil: +49 (0) 160 2237603

 

Amtsgericht Hamburg HRB 167883

UST-ID: DE304113771

WEEE: 24853534

 

www.conucon.de

 

Closed #231 <https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/231> .

petervanderwalt, 2022-01-17:
> I understand that, but would it not be possible to let CONTROL communicate on itself via a RS485 dongle with the VDF ?

CONTROL fronts the firmware, firmware controls the spindle.  We prefer to let CONTROL do just that - let the firmware do its job. Adding external control loops is dangerous and not something we intend supporting. 

0-10v based solutions works perfectly. RS458 gives you zero additional features, just complexity. See https://docs.openbuilds.com/doku.php?id=docs:blackbox:connect-vfd

conucon, 2022-01-19:
Hi Peter,

 

thank you. Yes, I know the Truth Coffee. My wife is from here. So we will be back hopefully next year.

 

Best reagds,

Felix

 

CONUCON UG (haftungsbeschränkt)

 

Billhorner Brückenstraße 40

20539 Hamburg 

Deutschland

 

Geschäftsführer: Dr. Felix Oehme

Email: ***@***.***

Tel.: +49 (0) 40 756791670 00

Mobil: +49 (0) 160 2237603

 

Amtsgericht Hamburg HRB 167883

UST-ID: DE304113771

WEEE: 24853534

 

www.conucon.de

 

I am in Graaff Reinet, some 670km from Cpt - would've been nice, but its a little far away. Enjoy the Mothercity though! Stay safe!
If you want to know where the best coffee is: http://www.truth.capetown/

pixelmagic66, 2022-01-21:
@petervanderwalt 

> 0-10v based solutions works perfectly. RS458 gives you zero additional features, just complexity. 

Zero seems not correct, control via rs485 gives feedback about the current speed at the least if I remember correctly ? That seems to be usefull specialy with hight speeds (so I read, the CNC has not arrieved here yet :) )