OpenBuilds/OpenBuilds-CONTROL #236: Have to run openbuildscontrol command twice to get it to start up

bubbapizza, 2022-01-27:
This very "WTF??".  The only way I can get openbuilds control to start up is by first running the 'openbuildscontrol' command line, which does nothing but show the message below and hang:

(node:912573) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
statusCode: 200

THEN

I open another command prompt and type in 'openbuildscontrol' again and it starts up!!!!!???  What is going on here?

Mumfrey, 2022-01-31:
When you start the program it runs in the system tray, Windows is probably hiding the icon. If you run the program again and it's already running it makes the already running instance visible.

It's likely that all you need to do is customise your system tray icons to make the openbuilds control icon always visible.

petervanderwalt, 2022-02-04:
And on subsequent reboots the first start is already done so you'll never even notice it.

closed 2022-02-04 by petervanderwalt

bubbapizza, 2022-02-14:
Ohhh, ok, I was running it on Ubuntu and yes it does show up in the system tray in the top left corner.

Thx