OpenBuilds/OpenBuilds-CONTROL #240: Finish Modal parser

petervanderwalt, 2022-03-15:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/index.js#L2506-L2508 to be extended for all other modals as needed for one of  @sharmstr 's projects

petervanderwalt, 2022-03-29:
v1.0.314 has a couple more modal statusses available 

[image]

sharmstr, 2022-04-12:
Thank you for this.  Quick question though.  I was testing and issued a "g20" (notice lowercase g) and thought it didnt work.  The problem is that it doesnt switch units if you use a lowercase g instead of an uppercase G.  Is this a grbl thing or a Control issue?

[10:25:16] [ g20 ] ok  // notice that grbl only responds with ok, not a section line with current modals

[10:25:28] [ G20 ] ok

[10:25:28] [ $G ] [GC:G0 G54 G17 G20 G90 G94 M5 M9 T0 F0 S0]

petervanderwalt, 2022-04-12:
That might be us.  
Send a lowercase g20 and then manually a $G - to see if Grbl took it?

sharmstr, 2022-04-12:
Yep.  grbl accepts it.  All of my code uses capitals in gcode so its not a big deal, but you might want to sort that out eventually.  :)

sharmstr, 2022-04-12:
Ok.  So if I issue a g20, I believe grbl accepts it.  I dont know until I then issue a $G to check.   The only reason I say this is because if I issue a g20 and check laststatus, it will still say G21.  BUT if I then issue a $G, then laststatus will update and correctly show G20.

petervanderwalt, 2022-04-13:
Cool then - fix incoming soon (check for modal commands regardless of case, to send $G after)

petervanderwalt, 2022-04-19:
[image]
V1.0.318 or later - out soon (:

closed 2022-04-19 by petervanderwalt