OpenBuilds/OpenBuilds-CONTROL #241: Alarm dialog box closes quickly

rlwoodjr, 2022-03-17:
If you home your CNC and it is not successful, the dialog box comes up and gives you information about the error, but only stays open for a second or two.  Should this stay open until you select the  to Clear Alarm or Cancel?

Dialog box is in the  websockets.js
socket.on('toastErrorAlarm', function(data)

petervanderwalt, 2022-03-29:
Have you picked it up on anything other than homing alarms?

petervanderwalt, 2022-03-29:
And you are running stock standard Grbl right?

petervanderwalt, 2022-03-29:
Did a quick little test, we clear all outstanding errors when Grbl restarts.   

Seems grbl reboots itself after some homing errors (that'sm weird) - I eliminated CONTROL by testing in a serial terminal (putty):

Home (on a machine with switches disconnected)
Gets alarm 9
Reboots itself (tested with both regular Grbl and GrblHAL)
[image]

Not sure why Grbl would do that - but i guess safety first kind of thing.  We unfortunately need the "clear all old faults" on reset for a lot of other reasons so will have to think how to work around this particular one...   It's actually weirder that Grbl reboots...

petervanderwalt, 2022-03-29:
Workaround implemented in v1.0.314 - will give you a code snippet later as its part of a bigger commit

rlwoodjr, 2022-03-29:
Interesting. I am not running stock grbl, but I don't think any of the changes will cause the reset. I will look into it,

I have not noticed on any other errors except homing.

I will look forward to the code snippet and version 1.0.314.  

Thanks for the help.

petervanderwalt, 2022-03-29:
Exclusions: https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/1a4dad34b00038103468569ffeb9a80c14a45b1e/app/js/websocket.js#L386-L388 

Basically not adding those alarms to the list to be closed by https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/1a4dad34b00038103468569ffeb9a80c14a45b1e/app/js/websocket.js#L424-L431 which runs when Grbl resets (and some other conditions).

closed 2022-03-29 by petervanderwalt