OpenBuilds/OpenBuilds-CONTROL #242: enhancement: Location Display for G53 machine coordinates

itchytweed, 2022-03-19:
First off, the program is great in the way that it passes g-code expressions on to the controller rather than edit them out or try to act in a non-standard way. I am currently using grblHAL with OpenBuilds-CONTROL. My request would be to include the G53 machine location values on the main display along with the current work locations display currently present. For example:

    MACHINE        G54 WORK
X   -245.551           10.021
Y   -532.328           13.558
Z     -37.082            -2.234

If not a concurrent display, how about an option tab where it says to display either in mm or inches to show machine coordinates in either inches or mm. Tabs could read "Machine - in", "Machine - mm", "Work - in", Work-mm".  The reason behind this because some programmed routines rely on absolute machine location exclusive of current work universe, such as Z-setters, tool change locations, safe positioning, etc. 

Thank you for a simple program that does its intended task quite well.

petervanderwalt, 2022-03-29:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/issues/170 relates and has a screenshot

closed 2022-03-29 by petervanderwalt

itchytweed, 2022-04-02:
@petervanderwalt - Thank you for the assist and understanding the design philosophy. Still a great program to use with grblHAL on the Lead 1010.