OpenBuilds/OpenBuilds-CONTROL #247: Dont resize window when sending code to 3001/upload

sharmstr, 2022-05-14:
Small annoyance and another "me" issue.  I never noticed it because I dont use OB Cam, but when posting gcode to Control via https://mymachine.openbuilds.com:3001/upload the window resizes to default size.   I run fullscreen on my machine.

I notice it now because the my tool change macro allows the user to select multiple files as a 'project'.   It will auto load the next file for them after a tool change.  I'm using https://mymachine.openbuilds.com:3001/upload to do it.  Maybe there's a better way.

petervanderwalt, 2022-05-24:
No cool! Good to have someone else test something - even just "to know" (:

closed 2022-05-24 by petervanderwalt

sharmstr, 2022-05-24:
Figured it out.  It happens when you have Control taking up half of the screen.  I have a extra wide monitor on my desktop where I do testing.  I had Control taking up the left half.  If you transfer, Control will resize to the default.  It however, doesnt happen when its in fullscreen.  That's all I care about so no need to track it down.

petervanderwalt, 2022-05-26:
Upon uploading we call browserWindow.show() https://www.electronjs.org/docs/latest/api/browser-window#winshow - doesn't seem like is should change the size, should just bring it to front. Weird one