OpenBuilds/OpenBuilds-CONTROL #248: Automatic probe socket doesnt turn off when probe doesnt finish

sharmstr, 2022-05-15:
This is a strange one that is another me issue, but thought you should be aware.

If you initiate automatic probe and click "stop" (maybe even abort) before it finishes, the prbResult socket is still on.  It doesnt seem to be a problem for the stock probing routines, however the console will still show the "steps" in the console .  

Here's some log info:

//Start Automatic Probe
[20:51:44] [ G10 P0 L20 X0 Y0 Z0 ] ok
[20:51:44] [ G0 X22.5 Y22.5 ] ok
[20:51:46] [ ] ok

//Stopped Automatic Probe

//Run stock Z Probe
**(console log)
Step 1 {x: "-77.498", y: "-176.966", z: "-50.171", state: "1"}**

//Run  stock Z Probe Again
**(console log)
Step 2 {x: "-77.498", y: "-176.966", z: "-48.368", state: "1"}**

For reasons I havent figured out yet, if I initiate my probe routine, my probe will finish but then Control tries to finish the previously stopped automatic probe routine.  I've worked around this by issuing a socket.off('prbResult') before my probe runs.

//Run My Tool Probe
**(console.log)
Step 4 {x: "-99.999", y: "-200.000", z: "-47.599", state: "1"}**

My probe makes contact, then the machine tries to finish the automatic probe.  I get g-code lockout errors when my code tries to issue a G10 since the machine is moving to finish the auto probe.  Then I get probe failed errors.

(console log)
**Probe Failed**
websocket.js:406 error: 9 - G-code locked out during alarm or jog state [ G10 P0 L20 Z9.060 ]<hr>This error may just be a symptom of an earlier event:<br> ALARM: 5 - Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.
websocket.js:406 error: 8 - Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job. [ $J=G91G21Z5F1000 ]
websocket.js:406 error: 9 - G-code locked out during alarm or jog state [ G21 ]

(serial log)
[21:00:32] [ [ PROBE ] ] Probe Completed.  **- THIS IS MY PROBE ROUTINE**
[21:00:32] [ G38.2 Z-50 F100 ] [PRB:-99.999,-200.000,-47.599:1]
[21:00:32] [ G38.2 Z-50 F100 ] ok
[21:00:32] [ G4 P0.4 ] ok
[21:00:32] [ G4 P0.3 ] ok
[21:00:32] [ G91 ] ok
[21:00:32] [ $G ] [GC:G38.2 G54 G17 G21 G91 G94 M5 M9 T0 F100 S0]
[21:00:32] [ $G ] ok
[21:00:32] [ G0 X-1 ] ok  **THIS IS WHERE IT ATTEMPTS AUTO PROBING** 
[21:00:32] [ G90 ] ok
[21:00:32] [ $G ] [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F100 S0]
[21:00:32] [ $G ] ok
[21:00:43] [ ] ALARM: 5 - Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4. [ G38.2 X-10 F50 ]
**[21:00:43] [ G38.2 X-10 F50 ] ALARM:5**
[21:00:44] [ [ PROBE ] ] Probe move ERROR - probe did not make contact within specified distance
[21:00:44] [ G38.2 X-10 F50 ] [PRB:-99.999,-200.000,-47.599:0]
[21:00:44] [ G38.2 X-10 F50 ] ok
[21:00:44] [ ] error: 9 - G-code locked out during alarm or jog state [ G10 P0 L20 Z9.060 ]<hr>This error may just be a symptom of an earlier event:<br> ALARM: 5 - Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.
[21:00:44] [ ERROR ] error: 9 - G-code locked out during alarm or jog state [ G10 P0 L20 Z9.060 ]
This error may just be a symptom of an earlier event:
ALARM: 5 - Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.
[21:00:44] [ G10 P0 L20 Z9.060 ] error:9  **THIS IS MY CODE TRYING TO G10**
[21:00:44] [ ] error: 8 - Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job. [ $J=G91G21Z5F1000 ]
[21:00:44] [ ERROR ] error: 8 - Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job. [ $J=G91G21Z5F1000 ]
[21:00:44] [ $J=G91G21Z5F1000 ] error:8
[21:00:44] [ $G ] [GC:G38.2 G54 G17 G21 G90 G94 M5 M9 T0 F50 S0]
[21:00:44] [ $G ] ok
[21:00:44] [ ] error: 9 - G-code locked out during alarm or jog state [ G21 ]
[21:00:44] [ ERROR ] error: 9 - G-code locked out during alarm or jog state [ G21 ]
[21:00:44] [ G21 ] error:9
[21:00:44] [ $G ] [GC:G38.2 G54 G17 G21 G90 G94 M5 M9 T0 F50 S0]
[21:00:44] [ $G ] ok
[21:00:46] [ ] Grbl 1.1g ['$' for help]
[21:00:46] [ ] [MSG:'$H'|'$X' to unlock]
[21:00:47] [ ] [MSG:Caution: Unlocked]
[21:00:47] [ ] ok

sharmstr, 2022-05-16:
I did more testing and even the stock probes have issues.  I'll DM you a link to some video evidence :)

The fix might just be as simple as issuing a socket.off('prbResult') when the probe wizard opens.   However, that might not be enough incase users have custom probe macros.

For my stuff, I issue a socket.off('prbResult') before I do any probing, but I also check for alarm state or if stop button was clicked during stock probing routines.

petervanderwalt, 2022-05-16:
Agreed, definately need to cleanup the socket listener for prbResult when its cancelled!   Good catch!