OpenBuilds/OpenBuilds-CONTROL #249: error: 1 - G-code words consist of a letter and a value. Letter was not found.

Mobiletainment, 2022-06-05:
I'm filing this issue for a friend who is not familiar with GitHub, but I hope we can still troubleshoot this issue here.

## Tools:

He's using *[Shape3d with the CNC mode](http://www.shape3d.com/support/User_Manual_V9.htm#_Toc454958733)* and the *OpenBuilds CONTROL - Machine Interface Controller*

## The issue

Shape3d produces a `.NC` file with the generated G-code. It's possible to load it into the controller software and run a simulation.
Nevertheless, if you actually start the process, the following error appears:

```
Grbl Error:
error: 1 - G-code words consist of a letter and a value. Letter was not found. [ [ New model Wed Jun 1 17:10:59 2022 ] ]
```

[image]

---

I'd appreciate to get some hints on how to fix this.

Thank you.

sharmstr, 2022-06-05:
He need to use a grbl compatible post processor.  Tell him to ask Shape3d if they have one. The "New Model" line is a comment and needs to be prefaced with a semi colon.  

More details in the grbl wiki:  https://github.com/gnea/grbl/wiki

Mobiletainment, 2022-06-06:
Thank you @sharmstr, without even getting the chance to look at the file you got it all correct.

The meta-information - which was not marked as a comment - was exactly the culprit.

The quick fix works:
[image]

We'll get in touch with the supplier to talk about the GRBL-conform post-processing.

Thank you very much for the help!

closed 2022-06-06 by Mobiletainment