OpenBuilds/OpenBuilds-CONTROL #256: Enable self-documentation of 'unknown' $codes

ImagineerNL, 2022-09-09:
[image]

A lot of the red grbl codes are known by my or my controller documentation.
The controller itself returns them as ; unknown
Would be a great feature to be able to edit those lines, either in the app itself, or updating the backup.txt with the correct text after the ;

ex: $123=200.000  ;  A/Y2-axis acceleration, mm/sec^2

petervanderwalt, 2022-09-22:
The problem with those are, that even the GrblHAL wiki doesn't tell me what those are.  If you find some, and can point me at a) title b) description c) expected input range we can add them to the template.

petervanderwalt, 2022-09-22:
Thanks, however note only Grbl is officially supported. We won't be adding ability to edit descriptions at this time, we will however try adding all official Grbl descriptions that may be missing. Thanks

closed 2022-09-22 by petervanderwalt