OpenBuilds/OpenBuilds-CONTROL #258: High CPU load with just opening to GUI

JiMcNulty, 2022-09-15:
[image]

I know, its PI, not supported and such
But just reporting so you would know...

petervanderwalt, 2022-09-22:
Troubleshooting tab > Application Diagnostics > Disable High CPU users like the 3D view etc.

closed 2022-09-22 by petervanderwalt

JiMcNulty, 2022-09-22:
> Troubleshooting tab > Application Diagnostics > Disable High CPU users like the 3D view etc.

Thanks, It helped a little but the CPU usage is still unreasonably high (gui is idle, no file is open, fkms enabled):
[image]

For example, this is the CPU usage with an open chromium on youtube:
[image]

Something else is going on I guess...

petervanderwalt, 2022-09-26:
[image]

JiMcNulty, 2022-09-26:
> [image]
Man come on you know that your comment is not helping
We are talking about linux on a PI 4B 8GB with 3D drawing disabled....

petervanderwalt, 2022-09-26:
> PI 4B

Well, that's not officially supported, so please don't even open issues for Pi please. See https://github.com/OpenBuilds/OpenBuilds-CONTROL/wiki/Installing-on-Pi4 and the bold note right at the top there.

JiMcNulty, 2022-09-26:
> I know, its PI, not supported and such But just reporting so you would know...

So ya... 
You should value more users who take the time to report stuff to try to try and help