OpenBuilds/OpenBuilds-CONTROL #266: Q: How to contribute to the Wiki?

poWer4aiX, 2022-10-28:
I might be a fool, but how to contribute to the Wiki? I would like to bring some light into how to build the software from scratch. I see that I can clone it locally, but this seems not to be linked to the original repo...

I can also contribute to the README.md, but I guess the Wiki is a better place.

petervanderwalt, 2022-10-28:
I'm not sure, but does https://docs.github.com/en/communities/documenting-your-project-with-wikis/adding-or-editing-wiki-pages work for you?

Just keep in mind content may be moderated to comply with OpenBuilds internal/external policies and requirements.

poWer4aiX, 2022-10-28:
This is the explanation on how to maintain a Wiki, where you have write access to. I don't know on which levels you can define who can directly contribute to the Wiki. Of course, opening it for the world is not an option.

One option might be to add a separate repo which holds a copy of the wiki pages. This enables us to fork and contribute. But I don't know how to syncing the repo then with the wiki...

poWer4aiX, 2022-10-28:
Next option might be to stick with .md files in the main repo...

petervanderwalt, 2022-11-01:
Yeah unfortunately they don't provide more granular control it seems:

[image]

Either public, or full push priviledge.  

I guess submitting in an Issue is an option - it understands Markdown

closed 2022-11-25 by petervanderwalt