OpenBuilds/OpenBuilds-CONTROL #269: NodeJS version 14 the serialport npm package

betteroutthanin, 2022-11-13:
Seems that the pi-install.sh has to be updated to pull down a later version of NodeJS.

Current = 14.x, Should be 19.X

This will prevent the serialport package from throwing a heap of issues related to promises.

updated lines 14/15
`
echo "(5/10) Installing NodeJS 19.x..."
curl -sL https://deb.nodesource.com/setup_19.x | sudo -E bash -
`

petervanderwalt, 2022-11-25:
Correct, please confirm you've tested the change? If possible, submit a PR? (: https://github.com/OpenBuilds/OpenBuilds-CONTROL/pulls

petervanderwalt, 2022-12-05:
Implementing as an untested change in V1.0.339 or later to close the ticket - if it breaks something for someone - refer the first paragraph of the Pi install Wiki (: and let us know what to change

closed 2022-12-05 by petervanderwalt