OpenBuilds/OpenBuilds-CONTROL #270: File open doesnt remember last folder used.

sharmstr, 2022-11-16:
As of .336 file open defaults to last folder used before upgrading.

How I confirmed.

Upgraded to .336.  
File open would always default to my download folder.
Reverted back to previous version.
Opened a file in a new location (/gcode/Camera Jig)
Upgraded to .336
Now file open defaults to /gcode/Camera Jig
Opened file from /gcode/cuttingboard
Quit control.
File open still defaults to /gcode/Camera Jig

sharmstr, 2022-11-16:
Seems to work ok when running from a browser window  localhost:3000.    Got the idea to check it from here:  https://github.com/electron/electron/issues/1780

petervanderwalt, 2022-11-18:
We do have the accept attrib on the input.  We upgraded from Electron 9 to 19 (long overdue) 
But I think (can't yet find confirmation) that Electron wants us to stop using the Chromium dialogs for things like this and implement native dialogs https://www.electronjs.org/docs/latest/api/dialog#dialogshowopendialogbrowserwindow-options

brrian11, 2022-12-01:
I came here to log this very issue & if useful to know, I'm experiencing the same thing in .336 and again in .338.

CiaranWhelan, 2022-12-13:
I came in here to log the same issue. I am experiencing the same problem with .336. I updated to .338 last night and had the same issue.

petervanderwalt, 2022-12-13:
https://discord.com/channels/745037351163527189/1052326878456848464 still no response to Bug report on Electron, trying discord

petervanderwalt, 2022-12-13:
> I came in here to log the same issue. I am experiencing the same problem with .336. I updated to .338 last night and had the same issue.

If the issue is still open, its still unresolved. Read upward, and you'll see its an upstream bug/change so not something we can fix ourselves.    Not a majorly important ticket either, one can make use on Windows favourites to avoid searching for folders: https://www.digitalcitizen.life/how-manage-use-favorites-windows-explorer/ - just one extra click

CiaranWhelan, 2022-12-13:
Not a problem. I will roll back to an earlier version. Working with
favourites will not do since I am working from many design folders.

*Regards,*
*Ciaran Whelan*
Mobile: 0458 811 910

petervanderwalt, 2022-12-20:
Beta test of Native File dialogs in 1.0.342 - let us know if you find any issues

closed 2022-12-20 by petervanderwalt