OpenBuilds/OpenBuilds-CONTROL #272: V1.0.336 Sluggish after Probing wizzard, fixed by minimizing and maximizing window?

ImagineerNL, 2022-11-18:
Windows 10 here, 
No other programs open besides Openbuilds Control.
Since the update to V1.0.336 I have had numerous sluggish responding jog commands.
I managed to pinpoint it to the software, not the controller. (xProV5)

This, because it happens even without sending the jogging commands to the controller.
See video for sluggish response.

https://user-images.githubusercontent.com/62603552/202744116-a6c5a3a8-6d88-4812-8f99-166f5cb9f75a.MP4

This is resolved immediately when minimizing/maximizing the app.

Last few times it seemed to start to become sluggish after running and closing the probe wizzard.

taskmanager in windows shows absolutely zero extra resources in use, not by openbuilds, not by any other program. Not before, during or after becoming sluggish.
Its only this app; rest of the computer remains fully responsive.

Any ideas?

petervanderwalt, 2022-11-23:
> xProV5

What firmware are you running on it?

ImagineerNL, 2022-11-24:
The [june release](https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware) XYYZ_NC version.

But as I said; the sluggishness is also when not sending anything to the controller, as in the video its just mouse movement.

petervanderwalt, 2022-11-25:
No other reports yet, so unsure what the specifics are. Do you have another PC to test against?

ImagineerNL, 2022-11-25:
Had someone try it on his laptop and no issues. (has a blackbox controller, though)
For now i will consider it an issue on my own laptop, until i find more details.
Will close it for now and reopen when needed.

closed 2022-11-25 by ImagineerNL