OpenBuilds/OpenBuilds-CONTROL #283: FluidNC support

petervanderwalt, 2022-12-13:
[x] Basic comms
[x] platform ID
[x] Config dump
[ ] Config edit
[ ] Config presets for machines
[ ] Max travel parse for soft-limits jog
[ ] steps per mm parse for Calibration Wizards

petervanderwalt, 2022-12-15:
Todo Yaml parse into legacy $$ setting as used by other parts of CONTROL

grblParams.$30 : used by Tool On > Variable
grblParams.$110 : used by JogRate X
grblParams.$111 : used by JogRate Y
grblParams.$112 : used by JogRate Z
grblParams.$113 : used by JogRate A
grblParams.$22 : Used by jog
grblParams.$27 : Used by Goto Zero Machine Coordinates (pull off from switch)
grblParams.$20 : Used by Softlimits calculation
grblParams.$130 : Used by Softlimits calculation, Machine extents on 3D view
grblParams.$131 : Used by Softlimits calculation, Machine extents on 3D view
grblParams.$132 : Used by Softlimits calculation, Machine extents on 3D view
grblParams.$133 : Used by Softlimits calculation, Machine extents on 3D view
grblParams.$100 : Calibration: X
grblParams.$101 : Calibration: Y
grblParams.$102 : Calibration: Z

petervanderwalt, 2022-12-21:
Basic fluidnc config file https://github.com/bdring/FluidNC/pull/746/commits/4af173b928f6032263e434cb41de2b33d2fe0f43 made by @synman

synman, 2022-12-22:
I've created a wiki here:  https://github.com/synman/Octoprint-Bettergrblsupport/wiki/Hacking-the-OpenBuilds-BlackBox-X32

petervanderwalt, 2023-01-19:
https://github.com/bdring/FluidNC/pull/755 relates - but needs a couple adaptations to be more compatible with the typical OpenBuilds Machine kit

petervanderwalt, 2023-04-12:
https://openbuilds.com/threads/just-discovered-openbuilds-control-with-fluidnc.20338/#post-134242 

Disable "?" statusloop while dumping config file, fluidNC responds to realtime commands in the middle of the config dump

curieos, 2023-05-06:
I tried using the latest OBC with a machine running FluidNC today, and it went mostly well. I had two major things I was having issues with, one was connecting over USB. Not sure why, it sounded like that was working for this person based on the forum post:

> https://openbuilds.com/threads/just-discovered-openbuilds-control-with-fluidnc.20338/#post-134242

Connecting over the network worked surprisingly well. I'd have occasional delays while jogging using the web interface, but during the several hours using OBC, even on the mobile jog widget, I did not have any noticeable delay when jogging. No stuttering during running a program either. The OBC probing wizard also worked without issue, though I suppose that's to be expected, since FluidNC uses the same codes for that as GRBL.

The second issue I had was the "Home All" button in OBC was disabled, so I had to send $H in the console to get the machine to home. Any reason for this?

petervanderwalt, 2023-05-07:
> issue I had was the "Home All" button in OBC was disabled, so I had to send $H in the console to get the machine to home. Any reason for this?

Thats the above todo item ^

> * Config parse for CONTROL's "grblParams.xx" needed values (below)

We disable the button when homing is disabled in Grbl's settings. FluidNC's different config means we need to write code to parse its config and see if homing is enabled or not to update the button accordingly

curieos, 2023-05-20:
Do you have an ETA on when you think you'll be able to work on this? I downloaded the source and figured out how to build the app in preparation to take a stab at it, but I don't know how useful my contributions will actually be, especially if you plan to work on this more in the next couple weeks.

If it will be a while and you would like the help, could you give me some guidelines for how you want things implemented please?

curieos, 2023-05-27:
Sorry for not responding, I had a very busy week. Thanks for the info. I created a workaround for the oddity.

I also figured out why I was having difficulties connecting over USB. When booting, if wifi is enabled, the boot process takes longer since it connects to the network before spitting out the status message with grbl at the front. The first message it sends is `[MSG:INFO: FluidNC vX.X.X]`. Initially, I extended the connection timeout to get it to work temporarily, but it requires an excessively long timeout (around 15 seconds). I came up with a somewhat hacky solution that prevents the grbl socket from being emitted until the standard grbl message gets received when OBC detects the FluidNC intro message.

I also got config parsing functional. I don't think it's the cleanest, so I'm going to take another pass at it before I make that live.

curieos, 2023-06-03:
I pushed my changes. I rewrote the parsing code to use lodash since it handles failing to find properties nicely. There's definitely more work to do, but I think I've gotten it to a minimum viable level for my purposes. I'll put some more work into managing the config in OBC when I get some time.

rhinodavid, 2023-12-01:
Team --
I've tried about a half-dozen Gcode senders with FluidNC and CONTROL, as is, is hands-down the best.

The [FluidNC GRBL Gcode senders page](http://wiki.fluidnc.com/en/support/gcode_senders) unfortunately doesn't list CONTROL, but I'm going to try and get Wiki access and add it.

One issue I did notice:
The default FluidNC startup message reads like `Grbl 3.7 [FluidNC 3.7.10 (wifi) '$' for help]`, and CONTROL does not correctly parse it (I think it ends up using "Method 2", which I can't find in the code after a short look).

The log looks like:
```
[23:18:09] [ connect ] Detecting Firmware: Method 2 (Ctrl+X)

[23:18:09] [ ] Detected an unsupported version: Grbl .7.10. This is sadly outdated. Please upgrade to Grbl 1.1 or newer to use this software. Go to http://github.com/gnea/grbl

[23:18:09] [ ] Grbl 3.7 [FluidNC 3.7.10 (wifi) '$' for help]

[23:18:12] [ connect ] Firmware Detected: grbl version .7.10 on 192.168.0.55
```

Luckily, one can change the startup message.

Changing the startup string to `Grbl \V [\H]` will output `Grbl 3.7 ['$' for help]` and eventually you'll have a log line:

```
[23:23:04] [ connect ] Firmware Detected: grbl version 3.7 on 192.168.0.55
```

The startup message can be changed with the [$Start/Message](http://wiki.fluidnc.com/en/features/commands_and_settings#startmessage) command or via the GUI.
[image]

Hope this helps someone. -d

dJOS1475, 2023-12-01:
I found the easiest way to get Control working properly with FluidNC was to set the firmware to “ Grbl 1.1g”. Then it works great:

[image]
[image]

synman, 2023-12-01:
@dJOS1475 I may still have wiki access and can ask Bart about it if you'd like.

They are fairly active on Discord.

synman, 2023-12-01:
Something to keep in mind though with your work around... It may impact other senders that have been fully updated to support FluidNC.

I guess that isn't such a big deal if you plan on sticking with Control, but it is with mentioning for others.

viernullvier, 2024-01-25:
I've just updated my controller to the latest FluidNC build (3.7.13) and dug around in the OpenBuilds Control code - looks like we're almost there.

First, FluidNC has added support for a major chunk of legacy GRBL settings. Sending `$$` will return a config that is understood by OBC and will update the UI accordingly.

Second, I looked into a way to have OBC send the `$$` command directly after connecting. As it happens, this is the same behaviour as expected for GrblHAL, so I just set up my controller to impersonate GrblHAL:

[image]

Setting `Start/Message` to `GrblHAL \B (\R) \H` allows me to use the controller without issues, but of course it forgoes any FluidNC-specific functionality in OBC that might already be there. _**If someone reading this discussion just wants to get OBC v1.0.370 to work with FluidNC v3.7.13, this is the thing to change.**_

Here's my take on the remaining open tasks :

- **Config parse for CONTROL's "grblParams.xx" needed values:**
  This might no longer be needed since OBC can now just use `$$` in the same way as other firmwares.
- **Workaround for disabling queryloop during config dump*:*
  If OBC switches to `$$` config reporting for FluidNC, this workaround is no longer needed.
- **Config edit:**
  @petervanderwalt How intricate do you envision this to be? Should it be an editor like the current one for the classic GRBL config or would it be enough to let the ACE editor write the config back to the controller? If it's the latter, there are a couple things to keep in mind:
  - The output of `$CD` is _**not**_ an accurate representation of the config file according to the local file system. Some missing parameters are added with their default values, while comments are stripped out.
  - The actual config file can be retrieved via `$LocalFs/Show=config.yaml`.
  - It is possible to have multiple different config files on the same controller and switch between them. To check if a different config file is used, the active config file name can be fetched via `$Config/Filename` first.
- **Config presets for machines:**
  Turning the existing presets into a FluidNC YAML could just be as simple as a JS template literal. Of course, this will only be half the battle because FluidNC will also require a hardware configuration. OBC could however provide a base config at least for the X32 which can then be amended with the preset YAML.
- **xModem upload of config files** and **Add FluidNC to Flashing Tool:**
  Those two should just be a matter of doing the work.
- There's also one extra task that's already been discussed here, but not mentioned in the initial to-do:
  **The firmware version detection logic needs to be fixed.**
  OBC expects a string like `Grbl 3.6 [FluidNC v3.6.5 (wifi) '$' for help]`, but the `v` prefix has never been part of the default FluidNC version string. This leads to the issue that @rhinodavid described - the version string is offset by one character and therefore interpreted as an unsupported pre-1.1 GRBL build ([relevant code](https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/index.js#L1474-L1477)).  A proper, more stable solution would just use a regex to extract the major and minor version.

I'd be willing to take care of the following:
- Remove previous attempt of FluidNC YAML parsing, use config returned by `$$` instead
- Fix version detection, add warning when attempting to use a pre-3.7.13 version of FluidNC
- Add support for config backup/restore/save to firmware (hardcoded to `config.yaml` at first)
- Try to add support to the flashing tool if it is really as simple as @petervanderwalt described it

How does that sound?

dJOS1475, 2024-01-26:
Funny you mentioned GrblHAL, I’ve just ditched FluidNC for it. I was just not 100% happy with it. GrblHAL Has been rock solid.

https://youtu.be/u_wgNvjDEF4

NeOdYmS, 2024-04-01:
how can i contribute on this issue ?

petervanderwalt, 2024-04-08:
> First, FluidNC has added support for a major chunk of legacy GRBL settings. Sending `$$` will return a config that is understood by OBC and will update the UI accordingly.

That's very good news, the different settings thing needed a lot of work to accomodate, so that should ease adoption, I think? Reusing the Grbl Settings we have now (even with some modification, still friendlier than editing Yaml) 

> I'd be willing to take care of the following:
>
> * Remove previous attempt of FluidNC YAML parsing, use config returned by `$$` instead
> * Fix version detection, add warning when attempting to use a [...]

That sounds great! I am allocated on CONTROL for a couple days this week, so please help! Happy to Merge pulls, just don't have a lot of time to work on it myself (:

petervanderwalt, 2024-04-08:
> Any chance somebody can give me a code pointer to the GRBL detection logic. I've been string searching and can't seem to find it.
https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/8de3acec1f0d35e84072405a7bf02fdc2e3ea9eb/index.js#L1459-L1528

petervanderwalt, 2025-01-08:
Closing issue, FluidNC requests never come up, no one wants to help put in effort, so for now, closing. Always happy to accept new pulls though, feel free to reopen if you feel like contributing (:

closed 2025-01-08 by petervanderwalt