OpenBuilds/OpenBuilds-CONTROL #285: grblHAL Servo Support in CONTROL

petervanderwalt, 2022-12-19:
- [ ] Build BIN files to include in Wizards and Tools > Firmware Flashing Tool
- [ ] Enable UI's "Servo" buttons - in grbl-servo we had a response in [OPT:] but that won't be an option if grblHAL only needs PWM settings changed in $$, and no recompile 
- [ ] Other tasks

coldbender, 2022-12-20:
Hello,
I made some noise over in https://openbuilds.com/threads/blackbox-x32.19810/page-2#post-130492 about setting up a non-stock build of the MiniMill and a custom grblHAL. My request is low priority, as I have hacked the machine together to meet our needs.

I noted that the PWM signal only comes out of the 2-pin PWM terminal, and not out of the 3-pin hobby servo header. My guess is that this can be changed within ....../ESP32/Main/BlackBoxX32_map.h but I don't know which GPIO is wired to the Signal header pin. Would it be as simple as changing "#define SPINDLEPWMPIN         GPIO_NUM_25" to a different pin?

I could use the process of elimination by building grblHAL with every unused PWM-supported pin off the ESP32 https://www.upesy.com/blogs/tutorials/esp32-pinout-reference-gpio-pins-ultimate-guide but I didn't want to damage our board by doing something stupid. I also removed the lid of the X32 (which the fan is mounted to) but didn't see any obvious traces (and I'm even worse at Electrical Engineering than I am at Software!). It looks like only GPIO 1 and 3 are unused, but I wonder if they are reserved for serial comms.

petervanderwalt, 2022-12-20:
> edit: Disregard this comment! PWM definitely comes across both the PWM terminal and the 3-pin servo header.

Confirmed yes, PWM is directly wired:

[image]

petervanderwalt, 2023-02-01:
In V1.0.345 or later, if $33=50, the penup/pendown buttons are enabled.

petervanderwalt, 2023-05-04:
As per @terjeio suggestion https://openbuilds.com/threads/blackbox-x32-servo-connection.20395/#post-134769 
Should also set $34-36

From footnote 13 on https://github.com/grblHAL/drivers 
``` 13 Set $33=50 (PWM frequency), $34=5, $35=5 and $36=10 to generate a "standard" PWM signal: 20ms repetition rate, 1 - 2ms pulse length range.
 ```

commit 53b7f58d (2024-04-09): Fix #285
closed 2024-04-09 by petervanderwalt