OpenBuilds/OpenBuilds-CONTROL #288: Blackbox X32 OpenBuilds CONTROL flashing tool

Lerch-Adams, 2023-01-11:
The BlackBox 4X in OpenBuilds Control had a 2 axis with Z homing disabled under the flashing tool. That option is not present for the BlackBox X32 I am using now. Can the 2 axis with Z homing disabled be added for the BlackBox X32?

petervanderwalt, 2023-01-12:
Thanks for the report, will get it added urgently. In the meantime, you can customize grblHAL by compiling from source: https://github.com/grblHAL/ESP32 but if its too much - hang in there and we'll get the 2-axes firmware built and added soon.

kendaniels, 2023-01-12:
Just adding another vote/voice to the 2-axis firmware. Set my Acro 1510 up yesterday for a pen plotter project and am now stuck without 2-axis firmware. 

Support sent me here but I'm not sure this will do the trick? https://openbuilds.com/threads/blackbox-x32.19810/page-2#post-130492

Hopefully 2-axis firmware isn't too far away?

petervanderwalt, 2023-01-13:
> Support sent me here but I'm not sure this will do the trick? https://openbuilds.com/threads/blackbox-x32.19810/page-2#post-130492

That was for your about building a "pen plotter" - thus support refered you to the details of using a servo as the toolhead (: - seperate from 2-axes homing.  Lots of plotters are 3-axes. The servo is a tool (enabling/disabling the PEN, its not a Z axis)

petervanderwalt, 2023-01-13:
Updated ticket description to avoid future confusion

petervanderwalt, 2023-02-01:
Fixed in V1.0.345 or later

closed 2023-02-01 by petervanderwalt