OpenBuilds/OpenBuilds-CONTROL #289: grblHal allow wifi password to accept special characters

sharmstr, 2023-01-16:
I havent tested all special characters, but Control wont accept and exclamation point in the wifi password.  It will change 

password!

to 

password$$

petervanderwalt, 2023-01-16:
Try with $39=0  https://github.com/grblHAL/ESP32/issues/55

sharmstr, 2023-01-16:
Fair enough.  btw - changing $39 to 0 doesnt help.  I've tried setting with ioSender, built in via wifi and Control.  I guess I need to go change the password on 20 devices.  Lame.   But not a Control issue so I'll close.

closed 2023-01-16 by sharmstr

petervanderwalt, 2023-01-16:
> Fair enough. btw - changing $39 to 0 doesnt help. I've tried setting with ioSender, built in via wifi and Control. I guess I need to go change the password on 20 devices. Lame. But not a Control issue so I'll close.

Should we tag in Terjeio? Or would you please comment on the issue above?   Sure it can be fixed rather than the pain of changing wifi passwords on all devices

sharmstr, 2023-01-17:
I will.  Is there anything special about the compiled version we are running?