OpenBuilds/OpenBuilds-CONTROL #292: grbl settings tab doesnt load after F5 refresh

sharmstr, 2023-01-31:
Low priority issue .   Noticed that grbl tab doesnt display when connected to controller after an F5 refresh.  This happens via USB and WIFI.  Only tested on X32

petervanderwalt, 2023-02-01:
Did a quick test and it does display over here. X32 connected. Curious what might be different. 
Any clues in devtools console on refresh? 

[image]

sharmstr, 2023-02-01:
Just tested on my OG BB and it works as expected.  Its only happening on X32.

sharmstr, 2023-02-01:
Ok.  It happens after a firmware flash when "Flash firmware and erase settings" is selected.  It works correctly after assigning a machine profile.  So it seems to only happen when laststatus.machine.name = ''.  Hope that helps.

petervanderwalt, 2023-02-02:
> Ok. It happens after a firmware flash when "Flash firmware and erase settings" is selected. It works correctly after assigning a machine profile. So it seems to only happen when laststatus.machine.name = ''. Hope that helps.

That mustve been a hard pattern to pick up on - so well done! That would certainly help, as I would never have guessed. Replicated, but not dug in yet - will get on with it! (:

petervanderwalt, 2023-02-02:
OK, I could not replicate the I$ connection
But did find the bug in https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/f3bcf217de1d2d669ab88b41b8368cbbe06e6788/index.js#L627-L650 - the safetosend thing was some testing workaround for the Driverminder circuit on the old xPro that could hold Reset pin down on something like that. Commented it out for now, so it just emits the       io.sockets.emit('grbl', status.machine.firmware) upon connection (like from a refresh) > which in turn triggers showGrbl() in frontend

petervanderwalt, 2023-02-02:
Fixed in V1.0.345

closed 2023-02-02 by petervanderwalt

sharmstr, 2023-02-02:
Awesome!  I'll admit that I got lucky.  Thank you!