OpenBuilds/OpenBuilds-CONTROL #295: Remove Drivelist to fix Github Actions Compile (URGENT)

petervanderwalt, 2023-02-09:
https://github.com/balena-io-modules/drivelist/issues/420 relates, prevents us from running a release cycle. Need certification too, so need to use GA

petervanderwalt, 2023-02-10:
@rlwoodjr - yesterday I tested rolling back to drivelist 9.2.4 (which needs Node 14, so also rolled back to 14 in build yaml)
Has a successful build https://github.com/OpenBuilds/OpenBuilds-CONTROL/actions/runs/4138523785/jobs/7155029195 

Earlier commit that rolled back to node14: https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/7739c6f5d2f63d10bb4058461e4b7ce7ba30dd9b#diff-5c3fa597431eda03ac3339ae6bf7f05e1a50d6fc7333679ec38e21b337cb6721R22

Commit that rolled back drivelist to 9.2.4 https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/6903c45afdfc295ee311c8fe11f2800887d65727

Drivelist v11 is just too unstable - but we would love to be on Node 16 though.  Quick fix for now, I may fork drivelist @9.2.4 and update it to work with node16 when I get a chance

rlwoodjr, 2023-02-10:
Thanks, I will give that a shot, I have been watchng your progess.

petervanderwalt, 2023-02-13:
@rlwoodjr 

Running off our fork of Drivelist 9.2.4 and using Node 16 to Build

Updated package.json 

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/bfedaec67d84403e814de2ff35286f995474c7cd/package.json#L18

Updated build.yml

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/bfedaec67d84403e814de2ff35286f995474c7cd/.github/workflows/build.yml#L22

@sharmstr:
``` Figured it out. I downgraded to node 14 instead of 16.  ```

@sharmstr - you can also check if you can build locally using Node16 again, back working on Github Actions so high hopes

sharmstr, 2023-02-13:
Pulled your changes.  Had issues with install since I was running Python 3.11.  Installed 2.7, told npm to where to find it.  Deleted node_modules.  Ran npm install without any issues besides typical warnings.  Rebuilt electron.   Tried to run and got an error saying it couldnt find drivelist\js\index.js

```PS C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL> ./node_modules/.bin/electron-rebuild
⠏ Building module: bindings, Completed: 0Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
⠙ Building module: bindings, Completed: 0  serialport.cpp
⠋ Building module: bindings, Completed: 0  serialport_win.cpp
⠇ Building module: bindings, Completed: 0  win_delay_load_hook.cc
⠙ Building module: bindings, Completed: 0     Creating library C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\@serialport\bindings\build\Releas
  e\bindings.lib and object C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\@serialport\bindings\build\ 
  Release\bindings.exp
⠹ Building module: bindings, Completed: 0  bindings.vcxproj -> C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\@serialport\bindings\build\Releas
  e\\bindings.node
⠧ Building module: drivelist, Completed: 2Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
⠋ Building module: drivelist, Completed: 2  drivelist.cpp
⠧ Building module: drivelist, Completed: 2  device-descriptor.cpp
⠼ Building module: drivelist, Completed: 2  list.cpp
⠴ Building module: drivelist, Completed: 2  win_delay_load_hook.cc
⠧ Building module: drivelist, Completed: 2     Creating library C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\drivelist\build\Release\drivelist
  .lib and object C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\drivelist\build\Release\drivelist.exp 
⠇ Building module: drivelist, Completed: 2  drivelist.vcxproj -> C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\drivelist\build\Release\\driveli
  st.node
✔ Rebuild Complete
PS C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL> node_modules/.bin/electron .

App threw an error during load
Error: Cannot find module 'C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\drivelist\js\index.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (node:internal/modules/cjs/loader:360:19)
    at Module._findPath (node:internal/modules/cjs/loader:573:18)
    at Module._resolveFilename (node:internal/modules/cjs/loader:926:27)
    at n._resolveFilename (node:electron/js2c/browser_init:245:1105)
    at Module._load (node:internal/modules/cjs/loader:785:27)
    at c._load (node:electron/js2c/asar_bundle:5:13343)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\index.js:45:19)
    at Module._compile (node:internal/modules/cjs/loader:1118:14)
Error: Cannot find module 'C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\node_modules\drivelist\js\index.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (node:internal/modules/cjs/loader:360:19)
    at Module._findPath (node:internal/modules/cjs/loader:573:18)
    at Module._resolveFilename (node:internal/modules/cjs/loader:926:27)
    at n._resolveFilename (node:electron/js2c/browser_init:245:1105)
    at Module._load (node:internal/modules/cjs/loader:785:27)
    at c._load (node:electron/js2c/asar_bundle:5:13343)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (C:\Users\sharmstr\Documents\github-repos\OpenBuilds-CONTROL\OpenBuilds-CONTROL\index.js:45:19)
    at Module._compile (node:internal/modules/cjs/loader:1118:14) {
  code: 'MODULE_NOT_FOUND',
  path: 'C:\\Users\\sharmstr\\Documents\\github-repos\\OpenBuilds-CONTROL\\OpenBuilds-CONTROL\\node_modules\\drivelist\\package.json',
  requestPath: 'drivelist'```

sharmstr, 2023-02-14:
Also fails if I use Node 14

petervanderwalt, 2023-02-14:
> Deleted node_modules. Ran npm install without any issues besides typical warnings.

Try delete node_modules and the package-lock.json file then npm install? Sometimes the lock keeps the old versions

petervanderwalt, 2023-02-22:
Updating to     "electron-builder": "^v24.0.0-alpha.13" seems to have fixed the compile issues we had. Got rid of Drivelist anyway, but might bring it back now that the root cause was found

closed 2023-02-22 by petervanderwalt