OpenBuilds/OpenBuilds-CONTROL #314: Enhancement request

SW0rden, 2023-05-04:
As stated in the linked post, if there could be some enhancements (read fixes) that would avoid the issue described, it would be great.   Perhaps having OB Control revert to its initial start up mode after a job that has a mode change is completed would be the best, but the other issues with the drop down commands and the pre-reading of the job file and 3D viewer are part of it as well.

When it comes to the drop down list commands for each axes and the XYZ combined, there really should only be the options of GOTO machine coordinate 0 for each individually or all (G53 G0Z0 first),  and work coordinate 0 for X or Y, or X and Y with a machine 0 for the Z first, so that the Z axis is up all the way and clear of any possible obstructions and leave it there.  I see no good from sending the Z to work coordinate 0 or anything else like the 5mm.  What if you have zeroed to the spoilboard and the center of your part, and inadvertently do a GOTOZERO (work coord) (G0 Z5, G0 X0 Y0, G0 Z0), which is the first in the XYZ list?  First is a plunge that is likely below your material surface in this case, then a rapid to the center of it, and then another plunge ( if you haven't already crashed) to the spoilboard depth.  The second choice in that list also ends with the Z at WCS 0.

[https://openbuilds.com/threads/using-ob-control-in-g21and-running-job-in-g20.20398/#post-134733](url)

Guenni75, 2023-05-29:
Hello.
I also don't understand why you always have to move Z to 0. I see this again and again in many programs and in the beginning I broke a few milling cutters with the crap.

Is there a way to modify this Button??

petervanderwalt, 2024-04-05:
> I see no good from sending the Z to work coordinate 0 or anything else like the 5mm

Often used during manual toolchanges - returning to Work Zero, with endmill loose in collet touching stock, then tighten. Poor mans Toolchanges. 5mm option is for those without homing switches (stats still show about 20-30% userbase don't home) - 5mm is low, but prevents Z max crash on our small machines. 

> I also don't understand why you always have to move Z to 0.

Not sure what you mean, but all CNCs need homing to set machine coordinates, then Zeroing to set Work Coordinates. 

Closing issue,  thanks for the suggestions

closed 2024-04-05 by petervanderwalt

Guenni75, 2024-04-07:
Hi.
If you press gotozero all axes move to 0.
If there is any working Material on the machine, the Mill moves full speed into it!!!
X and Y Zero is ok, but Z???