OpenBuilds/OpenBuilds-CONTROL #320: Probe

housspin, 2023-06-02:
Just got the X32 and the interface, setup both and i have some issues
I cant use the probe with the interface, get an error.
[image]
[image]

housspin, 2023-06-02:
any idea? Thank you in advance.

petervanderwalt, 2023-06-03:
Checkout openbuilds.com/forums for technical support. Github is for bugs/development

Probe file not found = skipped Prepare USB step in Interface Documentation

Inverted probe state = forgot to load machine profile, or wiring incorrect, or spindle has a Ground loop fault

closed 2023-06-03 by petervanderwalt

housspin, 2023-06-03:
Thank you so much
Regards
Housspin

housspin, 2024-01-09:
Hi, OpenBuilds
Just purchased a VFD/spindle 1.5k air cooling system, and I need help to
connect the VFD to the Blackbox 32, i assume that with the VFD I can
control the Spindle via Blackbox32, any suggestions?
Thank you so much
Regards
Raul

petervanderwalt, 2024-01-26:
> Checkout openbuilds.com/forums for technical support. Github is for bugs/development

Checkout openbuilds.com/forums for technical support. Github is for bugs/development