OpenBuilds/OpenBuilds-CONTROL #321: Update CSC signing process for new HSM

petervanderwalt, 2023-06-07:
https://docs.digicert.com/en/digicert-one/digicert-keylocker/ci-cd-integrations/plugins/github-custom-action-for-keypair-signing.html

petervanderwalt, 2023-06-15:
@chscott will update here when we can proceed

petervanderwalt, 2023-11-07:
> Document https://docs.digicert.com/en/digicert-keylocker/ci-cd-integrations/script-integrations/github-integration-ksp.html really helps.
> 
> Here's how is our application signed by KeyLocker: [nervosnetwork/neuron#2913](https://github.com/nervosnetwork/neuron/pull/2913)
> 
> There are mainly two steps:
> 
> 1. Setup signing runtime: https://github.com/nervosnetwork/neuron/pull/2913/files#diff-170ebc8e4dc40acf23cbe0ecce5f3e2aef1652511f59860db704106b197e1d52R54-R85
> 2. Sign application: https://github.com/nervosnetwork/neuron/pull/2913/files#diff-f1a2ada293a9fd7da045908348b61a30018539ff94b2cf54461bd122f03736ccR13-R15

petervanderwalt, 2024-04-09:
OK so what worked in the end was to 

a) create our own custom signing script:

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/4800540ffaa517925fc2cff26670809efa341ffe/signWin.js#L1-L31

b) Setup the Digicert Keylocker tools per https://docs.digicert.com/es/software-trust-manager/ci-cd-integrations/plugins/github-custom-action-for-keypair-signing.html with some modifications. Did use the same Github secrets names etc

Final version of build.yaml (Github Action)

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/4800540ffaa517925fc2cff26670809efa341ffe/.github/workflows/build.yml#L22-L62

Then github action still calls electron builder as usual
https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/4800540ffaa517925fc2cff26670809efa341ffe/.github/workflows/build.yml#L75-L88

While it runs, it uses the winSign.js script by calling it out from the package.json > build > win section

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/4800540ffaa517925fc2cff26670809efa341ffe/package.json#L127-L130

electron-builder then uses the script to sign it using smtools (provided by digicert) and the certificate fingerprint - pulling it from keylocker using smtools.

closed 2024-04-09 by petervanderwalt

niudai, 2024-07-07:
Thanks Peter, will try it out according to this thread

praisethemoon, 2024-07-17:
To be honest, I was keeping a close eye on this thread hoping for a "thread of hope" on this subject 😅 .

What are your thoughts on Azure Trusted Signing? https://azure.microsoft.com/en-us/products/trusted-signing Do you think it is a good alternative?

petervanderwalt, 2024-07-18:
DigiCert fixed my login so I am working again, not sure what went wrong on their end

Azure not investigated, but please do give it a go if you want to, would love to find alternatives that work

praisethemoon, 2024-07-29:
Azure is not available for individuals (like me) at the moment, they claim it will be publicly available at the end of the summer. 
Once It does, i will give it a shot and get back to you :)

Cheers.

niudai, 2024-08-25:
Hi Peter, thank you again, just one another question.
I wanna buy Digicert EV certificate + Cloud Keylocker store by around $1000 per year.
Will digicert charge more around cloud code signing like SSL.com with it's esigner service?
I currently use esigner and their service is pretty pricy, they charge $100 per month for only 10 times of code signing per month...
Does Digicert has such limitations?
Or once I buy the Keylocker service and they will not charge more and allow me to sign any times?

petervanderwalt, 2024-08-27:
I am not sure, I just do the engineering, someone else in our company takes care of the billing. Sorry

DigiCert so far seems unlimited I've run 30-40 test builds all signed so I don't think its limited - but reach out to them, they have a great online chat thats always given me 5 star service

dereklance, 2024-09-20:
> Or once I buy the Keylocker service and they will not charge more and allow me to sign any times?

from what i've seen the keylocker cloud service allows you to sign 1000 times per year