OpenBuilds/OpenBuilds-CONTROL #323: Move Macros Issue with > 9 Macros

sharmstr, 2023-06-25:
Hey Peter, 
There's an issue with moving macros when there is more than 9 of them.  Instead of moving the macro it will duplicate it.

Check out the following video.

https://youtu.be/f7QdChPJMVs

petervanderwalt, 2023-06-27:
Good catch, weird one, I don't see anything obvious in the sorting code. Does it affect just the rendering, or also the stored macros in localStorage (Does an F5 refresh fix it?)

sharmstr, 2023-06-27:
It gets stored.  Wish I had time to dig into it.  As usual it really only effects me.  LOL.  Just leave this open and one of us will get to it eventually. 

Might have something to do with alpha sorting instead of numeric?   idk

sharmstr, 2023-06-28:
My guess was right.  Its a alpha sorting issue.  

In sortMacros functions, change

var indexes = [index, newIndex].sort(); //Sort the indixes

to 

var indexes = [index, newIndex].sort((a,b)=>a-b); //Sort the indixes

petervanderwalt, 2023-06-29:
opened so I remember to apply, thanks for finding the fix

closed 2024-04-07 by petervanderwalt