OpenBuilds/OpenBuilds-CONTROL #325: Request: Allow macros to detect the "ok" response from Grbl

ivomirb, 2023-06-30:
I am trying to implement responsive jogging with a joystick as described here: https://github.com/gnea/grbl/blob/master/doc/markdown/jogging.md

It requires that commands are queued immediately one after another.
Currently there is no way to hook onto the “ok” response after the command is processed by Grbl.
I have a proposed change in my fork of the source code: https://github.com/ivomirb/OpenBuilds-CONTROL/pull/6/files

My update also contains an optimization for addQToEnd to skip sending $G for jog commands since they don't affect the parser state. It improves the responsiveness of my jogging implementation.

Please consider this change for including in the official software.

sharmstr, 2024-04-07:
Can you reconsider adding emit('ok') please.  It does help with smooth jogging.

petervanderwalt, 2024-04-09:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/64536498a0ae110993565fca44cfad7605520029/index.js#L1543

Will that suffice?

petervanderwalt, 2024-04-09:
Is it rather the $G thats the issue?

closed 2024-04-15 by petervanderwalt