OpenBuilds/OpenBuilds-CONTROL #333: Remove DTR toggle to make CONTROL more compatible with native USB on grblHAL

petervanderwalt, 2023-09-18:


petervanderwalt, 2024-04-09:
@swarfer - referring to https://openbuilds.com/threads/openbuilds-control-software.13121/page-82#post-138238

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/4800540ffaa517925fc2cff26670809efa341ffe/index.js#L1098
We do have DTR disabled. Does it still not work on the Pico?

swarfer, 2024-04-11:
still not connecting
[18:44:48] [ connect ] PORT INFO: Port is now open: COM7 - Attempting to detect Firmware
[18:44:48] [ connect ] Checking for firmware on COM7
[18:44:48] [ connect ] Detecting Firmware: Method 1 (Autoreset)
[18:44:48] [ ] ok
[18:44:48] [ connect ] Detecting Firmware: Method 2 (Ctrl+X)
[18:44:49] [ connect ] Detecting Firmware: Method 3 (others that are not supported)
[18:44:49] [ ] error:9
[18:44:52] [ connect ] ERROR!: No supported firmware detected - See https://docs.openbuilds.com/doku.php?id=docs:blackbox:faq-usb-connection-failed for more details. Closing port COM7
[18:44:52] [ disconnect ] PORT INFO: Port closed

bCNC does connect to this device and reports
$I
[VER:1.1f.20230810:]
[OPT:VNSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,SED,RTC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[FREE MEMORY:227K]
[DRIVER:RP2040]
[DRIVER VERSION:230810]
[DRIVER OPTIONS:SDK_1.5.0]
[BOARD:https://github.com/Expatria-Technologies/PicoBOB]
ok

petervanderwalt, 2024-04-11:
[21:43:42] [ connect ] PORT INFO: Port is now open: COM7 - Attempting to detect Controller...

[21:43:42] [ connect ] Attempting to detect Controller (1): (Autoreset)

[21:43:42] [ ] ok

[21:43:43] [ connect ] Attempting to detect Controller (2): (Ctrl+X)

[21:43:43] [ ] GrblHAL 1.1f ['$' or '$HELP' for help]

I think the reset triggers the "GrblHAL 1.1f ['$' or '$HELP' for help]" that we keep an eye out for? 

https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/0f4f90931c8aee5f597390188f6fdee467a9d71b/index.js#L1455-L1471

Don't see that string in the bCNC output. Will add a or "[VER:1.1f..." check maybe?

petervanderwalt, 2024-04-19:
@swarfer added https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/0d76d6e37ae58eb218092c8eb4026269ba570cd0 

Can you please test with draft https://github.com/OpenBuilds/OpenBuilds-CONTROL/releases/tag/untagged-1bdd758d4fa6a06826bb when you have a minute

swarfer, 2024-04-20:
nope, Pico comes up as COM7 and CONTROL reports

[06:52:15] [ 3D Viewer ] WebGL Support found! success: this application will work optimally on this device!
[06:52:16] [ websocket ] Bidirectional Websocket Interface Started Succesfully
[06:52:17] [ update ] Checking for Updates
[06:52:17] [ update ] You are already running OpenBuilds CONTROL 1.0.372
[06:52:26] [ connect ] PORT INFO: Port is now open: COM7 - Attempting to detect Controller...
[06:52:26] [ connect ] Attempting to detect Controller (1): (Autoreset)
[06:52:26] [ ] ok
[06:52:26] [ connect ] Attempting to detect Controller (2): (Ctrl+X)
[06:52:27] [ connect ] Attempting to detect Controller (3): (others)
[06:52:27] [ ] error:9
[06:52:30] [ connect ] ERROR!: No Response from Controller - See https://docs.openbuilds.com/doku.php?id=docs:blackbox:faq-usb-connection-failed for troubleshooting information. Closing port COM7
[06:52:30] [ disconnect ] PORT INFO: Port closed

maybe I need to update the grlHAL on this device and retest.....
nope, that did not change anything

from bCNC a $I shows
$I
[VER:1.1f.20240416:]
[OPT:VNMSL,100,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,ES,EXPR,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:219K]
[DRIVER:RP2040]
[DRIVER VERSION:240408]
[DRIVER OPTIONS:SDK_1.5.0]
[AUX IO:2,0,0,0]

and it comes up in alarm state because there are no limits connected
after inverting all needed pins so it does not start in alarm CONTROL does not connect but reports error:2 instead of :9

swarfer, 2024-04-24:
i have now changed settings ($15 and $14 IIRC) so it does not boot in alarm state and now I see 'error:2' after connection fails.

bCNC still 'just works' though the terminal does now show much, just a $# and the result and some OK's

petervanderwalt, 2024-06-18:
Treated myself for Fathersday:  Will help debug the issue (: easier to have one on the desk
[image]

Fresh flash, all pin states wrong by default out of Web Builder @terjeio - might make initial config harder than it needs to be? 

```<Alarm|MPos:0.000,0.000,0.000|Bf:100,1023|FS:0,0|Pn:XYZHSEP>```

petervanderwalt, 2024-06-18:
Per we should actually toggle DTR not remove it https://github.com/grblHAL/RP2040/issues/13#issuecomment-1445378979

Did that, and can connect (: 

[image]

petervanderwalt, 2024-06-18:
https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/edcbaf1a71171b2d3e01753494311b05c8b7d795/index.js#L1244-L1264
Fixed, will be in V1.0.377 or later

petervanderwalt, 2024-07-16:
https://openbuilds.com/threads/v1-0378-update.21905/ messed it up again

petervanderwalt, 2024-07-19:
Fixed in 1.0.381 again, seems to work on BB4X, X32, Pi Pico, ESP32S3 in testing

closed 2024-07-19 by petervanderwalt