OpenBuilds/OpenBuilds-CONTROL #335: Keyboard shortcuts:  Handle MetroUI naming non-standards of some keys in Macros

sharmstr, 2023-09-26:
Macro keyboard shortcuts dont take into consideration MetroUI naming conventions.  For example, Control will correctly change arrowright to right when saving the macro.  But when you press right, it doesnt do the conversion from arrowright to right.  You'll get the "Macro not found for arrowright" error.

sharmstr, 2024-04-17:
But when you click right, its not printing out 'test' to the console.   You're getting an 'undefined' message.

sharmstr, 2024-04-17:
Not sure its the best place to fix it, but the searchMacro function doesnt match any of the special metro names.

`function searchMacro(prop, nameKey, myArray) {
  console.log(nameKey, prop, myArray)
  for (var i = 0; i < myArray.length; i++) {
    if (myArray[i][prop] === nameKey) {
      return myArray[i];
    }
  }
}`

For instance, nameKey will come be arrowright, but in the macro its right.  So it will never run the macro.

As a test, this works

`function searchMacro(prop, nameKey, myArray) {
  console.log(nameKey, prop, myArray)
  
  if (nameKey.toLowerCase() == 'arrowright') {
    nameKey = 'right';
  }
  for (var i = 0; i < myArray.length; i++) {
    if (myArray[i][prop] === nameKey) {
      return myArray[i];
    }
  }
}`

[image]

petervanderwalt, 2024-04-17:
Ahhhh now I see, sorry!

petervanderwalt, 2024-04-17:
Was already doing some of the conversion (alt/ctrl/shift) in keyboard.js https://github.com/OpenBuilds/OpenBuilds-CONTROL/commit/562da2f706e2a8869a2cf2614ca29bbaa7826d60#diff-e52cb2231950cdf92d7ef8b5c835919016ded13ae8bf679aac910c306d744147L135 
so borrowed https://github.com/OpenBuilds/OpenBuilds-CONTROL/blob/master/app/js/macros.js#L285-L302 and added it to the section in keyboard.js 

[image]

closed 2024-04-17 by petervanderwalt

sharmstr, 2024-04-17:
I did see that, but for some reason I thought it was in the bindKeys function and just wasnt working. Sorry I didnt catch that earlier. Thank you for the fix.