OpenBuilds/OpenBuilds-CONTROL #340: XYZ CONTROL

bubbatay, 2023-12-07:
After do a XYZ calibration, i tried to run a gcode. It started but then froze in place. I attempted to raise z and nothing. Then tried xy and nothing. the controller is connected. I can sart thed router.nothing. I've reflashed the firmware, restarted the computer, reloaded openbuilds software and still nothing. I am at a dead end and would like to hve your thoughts, I hope i didn't fry the controller but from what i did i do not know why it would be fried.

petervanderwalt, 2023-12-07:
Double  check your PSU and Grbl Settings (reload machine profile in case you changed any settings manually)

closed 2023-12-07 by petervanderwalt

petervanderwalt, 2023-12-11:
CUSTOM is the right place to start, combined with your old controller's backup.  Leave key values like $4, $5, $6, $10 set as per the profile though