OpenBuilds/OpenBuilds-CONTROL #345: Update grbl settings tab/form when settings changed in gcode or serial console.

sharmstr, 2024-04-01:
Hey Peter,

Continuing conversation from: 

https://openbuilds.com/threads/ask-grbl-cnc-questions-here.3000/page-53#post-143770

Recap:  I would like change the way grbl settings are handled when changed in gcode or via the serial console.  Currently, the grbl settings values in the grbl tab are not updated when the setting change happens outside of the form.  My original need for this was to be able to get the correct value from the grblParams[] variable.   I handled this by adding the following to websockets.js in the sockets.on('data') section.

`if ( /^\$\d*=/.test(data.command) && data.response == 'ok' ) {
      grblSettings(data.command)
    }
`

It allows me to get the correct value of grblParams and I noticed that it would help ensure the accuracy of your Firmware Compiled with Parameters section in the troubleshooting tab (if you ever decide to release it).  I can see it also being used for a potential test fire laser button (show/hide).

I'm trying to decide if this is sufficient enough or perhaps too broad.

petervanderwalt, 2024-04-01:
Can't wait till we can get a release out :( so many fixes its gonna be a big and good release when it finally happens again. 

Agreed, would be a good one to fix. Applying the command to the grblSettings directly (instead of rerunning $$) is a cool workaround I like it in essence

sharmstr, 2024-04-01:
Yeah, I cant wait for an update.  I have some working code for a plugin system that we might want to consider down the road.

Let me know if you need help with anything.

petervanderwalt, 2024-04-11:
Nevermind,

closed 2024-04-11 by petervanderwalt