OpenBuilds/OpenBuilds-CONTROL #382: Provide an option to download an update for the Control App.

gwcSoftware, 2024-12-14:
Please provide an option whether to download a new update for the Openbuilds Control app or remove the Download update progress message that is taking up space in the console window.

Any new download for an update seems to be very slow and providing an update status for every tenth of a % is useless anyway!

closed 2024-12-17 by petervanderwalt

petervanderwalt, 2024-12-18:
"Any new download for an update seems to be very slow" 
Check your internet, I am rural part of southern africa on a 3rd world mobile connection and full download takes about 2-3min here

"Please provide an option whether to download a new update" Updates are somewhat forceful - but the opposite is most folks click Cancel, then come to support and complain about issues long since fixed by an update. Sorry but no, we tried it before and the workload increase is not worth it. Folks just don't bother telling us "hey i skipped 12 updates but this and this is not working for me"

gwcSoftware, 2024-12-19:
Why are the update messages needed! They interfere with the other messages.