OpenBuilds/OpenBuilds-CONTROL #384: CAM integration without certificates

theor, 2025-01-05:
Quick question - I've been digging into the CAM commit log to find the defunct integration with CONTROL. my understanding is that the constant need for CONTROL updates is a consequence of CAM using the `https://mymachine.openbuilds.com:3001` url, hence needing up to date certificates. I've hacked CAM to use `http://localhost:3000/` instead and it works beautifully. any reason not to do that ? 
Thanks!

petervanderwalt, 2025-01-07:
Back in the day Chrome threw security errors when trying to use http and localhost - been about 7+ years so I don't have the specifics or a screenshot at hand anymore (also had other plans with it being a rememberable url we never got to as well). 

Dont know maybe they browsers aren't as uptight about it anymore? 

https://github.com/OpenBuilds/OpenBuilds-CONTROL/wiki/Developers:-API-to-Inject-GCODE-into-CONTROL

petervanderwalt, 2025-01-07:
Thinking a moment, but Is your hacked CAM running on localhost? (would explain why it then allows access to localhost)...
Not sure I recall the exact issue but i think now that I think about it, once deployed to https://cam.openbuilds.com chrome says hey the is running in a https environment no access to http anymore. Https://localhost cant be publicly signed. Self signed raises error. So add a url that resolves to localhost but allows it to be signed by lets encrypt etc

theor, 2025-01-07:
you're right, good point. IMO having the feature _only if you run CAM locally_ would be fair but I get it would be a pain to document/maintain

petervanderwalt, 2025-01-08:
Yeah if there was a way around, be sure we'd do it (less effort always a win) - there was no way around it, so we would have done what needed to be done (:

closed 2025-01-08 by petervanderwalt