OpenBuilds/OpenBuilds-CONTROL #396: Return to Zero - Wrong Order of Operation.

Durahl, 2025-06-26:
Greetings!

I was wondering if we could have a change in the Order of Operation how the Machine is being returned to Zero?

As of now - when issuing the Return to Zero Command from the Dropdown Menu - the Machine will **_FIRST_** move the Tool Head to `Z5` ( by either raising **_OR_** lowering it ) and **_THEN_** to `X0Y0` which - given most machines running OB CONTROL will be using Hold-down Clamps - is a reliable way to break a Tool as with the Tool first lowered / raised to just `Z5` before returning to `X0Y0` it is bound to collide with **_ANY_** Workholding Technique using a Clamping System as they're usually more than 5mm tall.

**_IMHO_** - Returning to Zero should **_FIRST_** raise the Tool Head all the way up to clear the Path for **_THEN_** returning to `X0Y0` to finish it with `Z0` ( or whatever ).

Durahl, 2025-07-02:
[image]
The `Goto XYZ Zero` Keyboard Shortcut is eternally bound to the offending `G0 Z5, G0 X0 Y0 Z0` Command which quite frankly should be removed **AND** being replaced with the `G53 G0 Z-1.000, G0 X0 Y0, G0 Z0` one as the new default.

There is no reason to ever risk using `G0 Z5, G0 X0 Y0 Z0` that put Endmills at **HIGH** risk colliding with Clamp Style Work Hold Solutions and **GUARANTEED** collisions whenever the Work Zero is situated on top of the Waste Board instead of the top of the Part.

**_PLEASE_** issue an Update to remove `G0 Z5, G0 X0 Y0 Z0` and replace it with the Tool save `G53 G0 Z-1.000, G0 X0 Y0, G0 Z0` to receive the Keyboard Shortcut Binding.

sharmstr, 2025-07-02:
You assume that everyone is using limit switches.  There are plenty of home built "openbuilds" machines out there that dont.  A simple solution for you is to write the macro that does what you want and assign a keyboard shortcut to it.  

[image]

Durahl, 2025-07-02:
The current default **_BREAKS TOOLS_**! 💢

The worst that can happen if you don't have a Limit Switch on an OpenBuilds "grade" Machine is the Stepper Stalling against the Frame and **_ANY_** Machine that might cause / take damage from such a stunt **_WILL_** either have Limit Switches or not be running OpenBuilds 😑